HAL-RO-Developer/iot_plat_doc #11: pythonの導入

makki0205, 2017-06-05:
python 3.5.2以上をインストールする

コマンドプロンプトでpython -Vをした時のスクショをここに貼る

```sh
python -V
Python 2.7.12
```

@Komei1009 
http://no-title.com/programming/python

hiyanaka, 2017-06-05:
```sh
C:\WINDOWS\system32>python -V
Python 3.6.1

C:\WINDOWS\system32>pip -V
pip 9.0.1 from c:\users\ohs50326\appdata\local\programs\python\python36\lib\site-packages (python 3.6)
```

teruya002, 2017-06-05:
[image]

teruya002, 2017-06-05:
```sh
C:\Users\teruya.taiki\Desktop>python -V
Python 3.5.2

C:\Users\teruya.taiki\Desktop>pip -V
pip 8.1.1 from c:\users\teruya.taiki\appdata\local\programs\python\python35\lib\site-packages (python 3.5)
```

RikuHashimoto, 2017-06-05:
```sh
Microsoft Windows [Version 10.0.15063]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Users\ohs50158>python -V
Python 3.6.1

C:\Users\ohs50158>pip -V
pip 9.0.1 from c:\users\ohs50158\appdata\local\programs\python\python36-32\lib\site-packages (python 3.6)

C:\Users\ohs50158>pip -V
```

yam123, 2017-06-05:
```sh
C:\Users\ohs50258>python -V
Python 3.6.1

C:\Users\ohs50258>pip -V
pip 9.0.1 from c:\users\ohs50258\appdata\local\programs\python\python36-32\lib\s
ite-packages (python 3.6)
```

show-t, 2017-06-05:
```sh
ohs50247@R73 MINGW64 ~/Desktop
$ python -V
Python 3.6.1

ohs50247@R73 MINGW64 ~/Desktop
$ pip -V
pip 9.0.1 from c:\users\ohs50247\appdata\local\programs\python\python36-32\lib\site-packages (python 3.6)
```

closed 2017-06-06 by makki0205