HAL-RO-Developer/iot_plat_doc #12: arduinoでjsonがParseできるか調査する

makki0205, 2017-06-05:
# 背景
データのやり取りにjsonを使うため
# 目的
jsonを簡単に扱いたい
# 対応内容
1. jsonをParseするについて調べる。
2. arduino言語でjsonをParseするライブラリを調べる。ここにまとめる
3. サンプルコードを書いてコミットする
# 期日
6/7 13:05まで

yam123, 2017-06-06:
１．終わりました。
２．https://github.com/bblanchon/ArduinoJson＜－のGithubからArdunoJsonが乗っています。

```
#include <ArduinoJson.h>
void setup() {
  Serial.begin(115200);
  while (!Serial) {
    // wait serial port initialization
  }
  StaticJsonBuffer<200> jsonBuffer;

  char json[] =
      "{\"data\":{\"time\":{\"day\":1,\"month\":3,\"year\":16,\"hours\":9,\"mins\":59,\"secs\":14}}}";

  JsonObject& root = jsonBuffer.parseObject(json);

  // Test if parsing succeeds.
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  int day = root["data"]["time"]["day"];
  // Print values.
  Serial.print("Day:");
  Serial.print(day);
}
void loop() {
  // not used in this example
}
```

yam123, 2017-06-06:
https://github.com/bblanchon/ArduinoJson　＜－のGithubからArdunoJsonが乗っています。

yam123, 2017-06-06:
https://github.com/bblanchon/ArduinoJson

makki0205, 2017-06-06:
問題なし 〆

closed 2017-06-06 by makki0205