HAL-RO-Developer/iot_plat_doc #6: ESPでSocket通信ができるように

makki0205, 2017-06-05:
ESPでTCPSocket通信をする

PCからESPにTCPSocketを用いて文字列を送る

hiyanaka, 2017-06-06:
[image]

makki0205, 2017-06-06:
ATモードで通信出来ても、あまり意味が無いのでコードで確実に送受信して、コードを挙げてください。

hiyanaka, 2017-06-07:
```c
#include <ESP8266WiFi.h>

const char* ssid = "MY SSID";  // SSID
const char* password = "MY PASSWORD";   // パスワード
const int LED = 16;

WiFiServer server(5000);  // ポート設定
WiFiClient client;

void connectWiFi(const char* ssid ,const char* password) {
   
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid ,password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }

  // WiFi接続完了
  Serial.println();
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());  // ESPのIPアドレスを表示
}

void setup() {
  Serial.begin(115200);
  pinMode(LED ,OUTPUT);

  connectWiFi(ssid ,password);
  server.begin();
  client = server.available(); 
}

void loop() {
   
  while ((WiFi.status() != WL_CONNECTED)) {
    connectWiFi(ssid ,password);
  }

  while (!client) {
    client = server.available();
  }

  while (!client.available()) { }
  Serial.println("client connected");
  
  String res = client.readStringUntil('\n');
  Serial.println(res);
  client.flush();
  
  if (res.equals("ON")) {
    digitalWrite(LED ,LOW);
  }
  else if (res.equals("OFF")) {
    digitalWrite(LED ,HIGH);
  }
  else {
    Serial.println("ERROR");
  }
}
```

hiyanaka, 2017-06-07:
[image]

makki0205, 2017-06-09:
これはesp受信だけですよね？
esp発信を実験してみてください

closed 2017-06-19 by makki0205