HAL-RO-Developer/iot_plat_doc #9: フルカラーLEDの調査及び選定

makki0205, 2017-06-05:
ステータスランプのフルカラーLEDの調査及び選定

1. 調査
2. 選定
3. 購入
4. 実装

作業がおひとつ終わるごとに何かしらの成果物をissuesのコメントに貼ってください。

teruya002, 2017-06-05:
項目1の調査については秋月電子のHPを見て調べました。
項目２の選定については5mmΦのフルカラーの中からICが内蔵されていない４本足のタイプとして見たところ１つしか種類がない状況です。
項目３についてはこの秋月のものにする場合最低でも２日は注文してから届くのに時間がかかる都合上明日中に購入する事はできません。
項目４については３が達成不能なため同じく明日することができません。
４つ一気に報告する形になってすみません。

makki0205, 2017-06-05:
購入予定のLEDのURLを貼ってください
それで２まで完了と致します

期日はそれを対応後決めます

teruya002, 2017-06-05:
http://akizukidenshi.com/catalog/g/gI-02476/
これです。

makki0205, 2017-06-06:
@RikuHashimoto が購入してくれたので受け取ってサンプルコードの作成を勧めてください。

makki0205, 2017-06-06:
## サンプルコードは
- led(red,green,blue);でledがフルカラーで表現できる関数を作成する
- 回路を写メでissueに貼る

teruya002, 2017-06-06:
中間報告
[image]
関数を使ってのフルカラーLEDは現在できていませんが、回路はできたので先に送っておきます

teruya002, 2017-06-07:
```sh
#include "fullLED.h"
void setup() {
  pinMode(9,OUTPUT);
  pinMode(10,OUTPUT);
  pinMode(11,OUTPUT);
}

void loop()
{
  void led(uint8_t red,uint8_t green,uint8_t blue);
{
  int red=9;
  int green=10;
  int blue=11;
  analogWrite(red,255);
  analogWrite(green,0);
  delay(1000);

  analogWrite(green,255);
  analogWrite(blue,0);
  delay(1000);
  
  analogWrite(red,0);
  analogWrite(blue,255);
  delay(1000);
}
}

//自作ヘッダファイル
#ifndef fullLED_H_INCLUDED
#define fullLED_H_INCLUDED

void led(uint8_t red,uint8_t green,uint8_t blue);

#endif
```
関数化のプログラミングと今回の授業で成功したフルカラー点灯の回路図です。

teruya002, 2017-06-07:
[image]

teruya002, 2017-06-07:
追記：上記の写真での抵抗の値はR:300Ω G:82Ω B:82Ωでやっています。

makki0205, 2017-06-09:
問題なし 〆

closed 2017-06-09 by makki0205