HAL-RO-Developer/iot_plat_doc #15: バッテリでマイコンを動かす

makki0205, 2017-06-06:
# 背景
マイコンをコードレスでも使えるようにバッテリを搭載する
# 目的
バッテリでマイコンを動かす
# 対応内容
1. 5vの小型リチウムバッテリを調査する(調査内容をここ記述)
    - 充電の可否
    - 充電の電圧
    - 最大の電流量
    - バッテリの容量
    - サイズ
    - 値段
2. マイコンの駆動電圧を調査
3. 選定を行う
    - 根拠をもとに選定する
4. 購入
# 期日
6/10まで

yam123, 2017-06-06:
[https://www.amazon.co.jp/%E3%82%BF%E3%83%9F%E3%83%A4-%E3%83%8B%E3%82%AB%E3%83%89%E3%83%90%E3%83%83%E3%83%86%E3%83%AA%E3%83%BC-7-2V-%E3%82%AB%E3%82%B9%E3%82%BF%E3%83%A0%E3%83%91%E3%83%83%E3%82%AF-55085/dp/B0019ECW02/ref=zg_bs_2395007051_9?_encoding=UTF8&psc=1&refRID=XAG639E4EBPRMZE36FFA](url)
タミヤ ニカドバッテリー 7.2V カスタムパック 55085
充電不可
7.2V
1300mAh
135×47×25mm
1,573円（元は2,376円）
重量約290g

今のところはこれがベストかなと思っています。大きさに関してちょっと大きのでコンパクトなバッテリーを探しています。

makki0205, 2017-06-06:
@yam123 駄目です。

yam123, 2017-06-07:
## パターン１
## バッテリー
- SYMA X5HW X5HC バッテリー 3.7V　500mah 3.7V ? 2個パック ※過充電保護機能付き [並行輸入品] 
- [](https://www.amazon.co.jp/dp/B01M3SRRVL?psc=1)
- 3.7V
- 500mAh
- 約8x24.5x44.5mm
- セット内容：バッテリーx2
- 1400円

## 充電
- 充電用回路参考
- [](https://strawberry-linux.com/pub/USB-uLIPO-REV-B.pdf)

## 充電用チップ
- マイクロチップ, バッテリー充電コントローラ IC, 3.75 → 6 V, 10-Pin MSOP
- [](https://jp.rs-online.com/mobile/p/battery-charge-controller-ics/7240523)
- 単価：122.60円x5 = 613円

## パターン2
## バッテリー
- LiPo電池 3.7V 1400mAh
- [](https://strawberry-linux.com/catalog/items?code=15078)
- 3.7V
- 1400mAh
- 8x34x50mm
- 1050円

## 充電モジュール
- USBバッテリーチャージャー USB-uLiPo
- [](https://strawberry-linux.com/catalog/items?code=15090)
- マイクロチップのMCP73833を使ったLiPo電池用充電器です。
- 30x13mm
- 600円

### バッテリー1個使用時は
NJM2360のICを使用し、電圧を昇圧する。
[](http://akizukidenshi.com/catalog/g/gI-02878/)
電源電圧範囲 (2.5V*～40V)
過電流検出回路内蔵
電源電圧 (V+ 2.5*V～40V)
出力電圧 (VOR 1.25V～40V)
2個で150円
### バッテリー2個使用時は
同じICを使用し、電圧を降圧する。
[](http://akizukidenshi.com/catalog/g/gI-02878/)

yam123, 2017-06-09:
# パターン２
## バッテリー
- LiPo電池 3.7V 1400mAh
- https://strawberry-linux.com/catalog/items?code=15078
- 3.7V
- 1400mAh
- 8x34x50mm
- 1050円

## 充電モジュール
- USBバッテリーチャージャー USB-uLiPo
- https://strawberry-linux.com/catalog/items?code=15090
- マイクロチップのMCP73833を使ったLiPo電池用充電器です。
- 30x13mm
- 600円

## 昇圧
- NJM2360のICを使用し、電圧を昇圧する。
- http://akizukidenshi.com/catalog/g/gI-02878/
- 電源電圧範囲 (2.5V*～40V)
- 過電流検出回路内蔵
- 電源電圧 (V+ 2.5*V～40V)
- 出力電圧 (VOR 1.25V～40V)
- 2個で150円

計算式　3.7V -> 5V
             3.7V -> 3.3V
http://www.njr.co.jp/products/semicon/PDF/NJM2360_J.pdf

TOTAL : 1650 + 150 + 300 = 2100円

closed 2017-06-12 by makki0205