HAL-RO-Developer/iot_plat_doc #17: EPROMデータ読み書き

makki0205, 2017-06-06:
# 背景
wifiや接続先サーバは電源が消えても記憶していてほしい。
# 目的
espでEPROMへのアクセスを自在に出来るようになる。
# 対応内容
1. espでEPROMへアクセスするライブラリを探す
3. サンプルコードを取得
2. 使い方を簡単にまとめる
# 期日
6/7

hiyanaka, 2017-06-07:
# 使い方
## インクルード
```c
#include <EEPROM.h>
```
## 簡単な例
### 初期化
```c
// 使用するサイズを宣言する
EEPROM.begin(100);
```
### 書き込み
```c
uint8_t val = 32; // 保存したい値

// 0 : EEPROM の先頭アドレス
EEPROM.write(0, val);

// コミット
// (コミットで内蔵フラッシュメモリに、実際に書き込みます)
EEPROM.commit();
```
### 読み出し
```c
// 0 : EEPROM の先頭アドレス
uint8_t val = EEPROM.read(0);
```
## 実用的な例
WiFi 接続のための、SSID と Password を保存する場合
```c
#include <EEPROM.h>

struct CONFIG {
  char ssid[32];
  char pass[32];
};

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println("");

  EEPROM.begin(100);

  CONFIG buf;
  EEPROM.get<CONFIG>(0, buf);
  Serial.println(buf.ssid);
  Serial.println(buf.pass);

  strcpy(buf.ssid, "Your SSID");
  strcpy(buf.pass, "Your Password");
  EEPROM.put<CONFIG>(0, buf);
  EEPROM.commit();
}

void loop() {
}
```
引用先：http://qiita.com/exabugs/items/0220edd4c9bf93da981b

makki0205, 2017-06-09:
問題なし　〆

closed 2017-06-09 by makki0205