HAL-RO-Developer/iot_plat_doc #19: ESP8266WiFi.hのAPIについて調査する

makki0205, 2017-06-06:
# 背景
ESP8266WiFi.hを使いこなす必要がある
# 目的
ESP8266WiFi.hについてのドキュメントをわかりやすく残す。
# 対応内容
1. ESP8266WiFi.hで用意されているメソッドを誰が見てもわかるようにissueに残す。
2. 各APIを全部網羅するプログラムを作成する。（可能な限り）
# 期日
6/6

RikuHashimoto, 2017-06-09:
Station class
Station classには、Wi-Fi接続の管理を容易にするためのいくつかの機能があります。接続が失われた場合、再度使用可能になると、ESP8266は最後に使用したアクセスポイントに自動的に再接続します。

Scan class
範囲内の利用可能なネットワークのスキャン機能およびリスト機能がある

UDP class
UDPメッセージの送受信を可能にします。 UDPは、データの完全性のチェックサムと、データグラムの送信元と送信先の異なる機能を扱うためのポート番号を提供します。

RikuHashimoto, 2017-06-12:
https://drive.google.com/drive/folders/0B8kiuN-BwfKBNUhkVnhZUXF0blk

RikuHashimoto, 2017-06-13:
メソッドについてまとめました
https://drive.google.com/drive/folders/0B8kiuN-BwfKBNUhkVnhZUXF0blk

RikuHashimoto, 2017-06-19:
今日、接続が切れた時に再接続するコード、電波強度やIPアドレスなどのパラメータをコンソールで表示するコードを作成し、コンソールの表示結果をコメントに残し、コードもgitに上げます

RikuHashimoto, 2017-06-19:
https://github.com/HAL-RO-Developer/iot_plat_doc/commit/c1af3a26eece5c948ba76c0fcc61baed5a19cd0e

makki0205, 2017-06-19:
スタックトレース出して
statusをレビュー待ちに変えてくださいね！

RikuHashimoto, 2017-06-21:
WiFi connected
IP address: 172.20.10.5
SSID: riku'siPhone
Password: Rikuh9212
RSSI: -33 dBm
BSSID: A2:BE:05:74:AE:50

https://github.com/HAL-RO-Developer/iot_plat_doc/tree/master/work/hashimoto/espwifiserial

RikuHashimoto, 2017-06-21:
Wifi再接続
https://github.com/HAL-RO-Developer/iot_plat_doc/tree/master/work/hashimoto/testWifi/restartWifi

closed 2017-06-21 by makki0205