HAL-RO-Developer/iot_plat_doc #20: デジタルPinからurat I/O

makki0205, 2017-06-09:
# 背景
デジタルピンでurat I/Oを実装する必要がある
# 目的
デジタルPinからurat I/Oについて調査が必要
どのライブラリを使うか。どのようなパラメータが必要か？
# 対応内容
- 以下のプログラムを作成する
    - マイコンを２つ用意しデジタルピンを用いて双方向通信を行う
    - 制御やライブラリ読みだしの行全てにコメントを降る
    - 通信にかかる時間を調査する。
    - 非同期にデータを受信できるかどうか調べる

- 以下についてまとめissuesに残す
    - 使うライブラリ
    - ライブラリのリファレンスの日本語化
    - 通信にかかる時間を調べる
    - 非同期処理の仕方
    - 
# 期日

hiyanaka, 2017-06-19:
https://drive.google.com/drive/u/0/folders/0B4FaZHuR9PHDWTY4QWtUYjNHZ2c

makki0205, 2017-06-20:
これは技術検証なので後でこのissueを見ればすべてが分かるようにしてください。
- ソースコードをどこに配置したか。
- どういう挙動のソースコードなのか
- ソースコードのトレースはる。
- 注意点や基本的な使い方。
- グーグルDriveよりもマークダウンでまとめてください。

hiyanaka, 2017-06-21:
- ソースコード
https://github.com/HAL-RO-Developer/iot_plat_doc/blob/master/work/nagaoka/ESP8266/ESP_Bidirectional01/ESP_Bidirectional01.ino

- 説明
マイコンを2つ用意し、送信側に指定した文字列を送ると受信側に表示

- トレース＆使い方
1.  ポート指定(RX,TX)
2.  ハードウェア、ソフトウェアシリアル初期化
3. 30文字以上もしくは、文末に『.』の文字列を受信したとき、受信側のシリアルモニタに表示する

hiyanaka, 2017-06-21:
# SoftwareSerial.hリファレンス
## SoftwareSerial(rxPin, txPin)
SoftwareSerial(rxPin, txPin)をコールすると、新しいSoftwareSerialオブジェクトが作成されます。下の例のように、そのオブジェクトに名前を付ける必要があります。SoftwareSerial.begin()を実行することも必要です。 複数のポートを同時に開くことができますが、受信できるのは1度にひとつのポートだけです。 
### パラメータ
rxPin: データを受信するピン 
txPin: データを送信するピン 
## SoftwareSerial.begin(speed)
シリアル通信のスピード(ボーレート)を設定します。サポートされているのは次の値です。 
300、1200、2400、4800、9600、14400、19200、28800、31250、38400、57600、115200 
### パラメータ
speed: ボーレート (long) 
## SoftwareSerial.available()
ソフトウェアシリアルポートのバッファに何バイトのデータが到着しているかを返します。
### パラメータ
なし 
### 戻り値 
バッファにあるデータのバイト数を返します。
## SoftwareSerial.read()
受信した文字を返します。同時に複数のSoftwareSerialで受信することはできません。listen()を使って、ひとつ選択する必要があります。 
 ### パラメータ
なし 
### 戻り値
読みこんだ文字 (データがないときは -1) 
## SoftwareSerial.print(data)
ソフトウェアシリアルポートに対してデータを出力します。Serial.print()と同じ機能です。 
### パラメータ
多くの種類があります。Serial.print()の項を参照してください。 
### 戻り値
送信したバイト数 (byte) 
## SoftwareSerial.println(data)
ソフトウェアシリアルポートに対してデータを出力します。Serial.println()と同じ機能です。 
### パラメータ
多くの種類があります。Serial.println()の項を参照してください。 
### 戻り値
送信したバイト数 (byte)

closed 2017-06-21 by makki0205