HAL-RO-Developer/iot_plat_doc #21: 3D物体を出力してみる

makki0205, 2017-06-09:
# 背景
3Dプリンターでプロトタイピングや制作を行う可能性があるため
# 目的
3D CAD及び3Dプリンターを扱えるようにしておく
# 対応内容
1. [こちら](https://drive.google.com/drive/folders/0B-XUZMi52MVvQWNpcW8yNUFRbkE)のデバイスimageをもとに3Dモデルを作成する(Fusion 360を使用)
2. 3Dプリンタで3Dモデルを出力する方法を調査する(先生に聞いて来る)
3. 3Dプリンタで出力する方法をまとめる
4. 必要な材料を揃える
5. 出力する
# 期日
９/17

teruya002, 2017-06-10:
中間報告
Fusion360ダウンロード完了

teruya002, 2017-06-15:
中間報告
3Dモデルを作成した際の現在各寸法
縦・・・54mm（残り96mm）
横・・・81mm（残り19mm）
高さ・・29mm（残り1mm）

teruya002, 2017-06-16:
交渉の件に関して
松山先輩に話をしてみたところ、先輩の名前の書いてある材料ならば使ってもいいとのことです。
容量としてはこちらがやろうとしてるものができるかもしれないとの事です。
値段交渉に関してはその話すらでてないです。

makki0205, 2017-06-19:
出力してみてください

teruya002, 2017-06-27:
[image]

手前の立方体は1辺 10mmです。
奥の大きい図形のほうの寸法
縦 54mm
横 85mm
高さ 20mm

四隅の角は
フィレットで各 5mm 指定です。
内側
縦 44mm
横 75mm
高さ 15mm
土台の足の太さは
約4.8mm～5.1mm

closed 2017-06-30 by makki0205