HAL-RO-Developer/iot_plat_doc #22: 何pin必要か調査する

makki0205, 2017-06-12:
# 背景
esp8266でpinが足りるか調査する必要がある
# 目的
全pin足りるかどうかの調査
# 対応内容
仕様を満たす電気配線のイメージ図を作成する
# 期日
6/14

yam123, 2017-06-14:
##必須PIN
- I2Cポート : PIN4とPIN5
- APモードスイッチ : PIN16
- ANALOGポート : TOUT (測定電圧は0V-1Vである点に注意)
- DIGITALポート : 2セット　（PIN0とPIN2  && PIN15とPIN13）
- 電源スイッチ : PIN 3.3v
- パイロットランプ : PIN 3.3V
- ステータスランプ（2色） : PIN12とPIN14

yam123, 2017-06-21:
## ESP-32
https://www.switch-science.com/catalog/3170/

Digital Port = 4 pin (x2 Port)
- DPort1 = pin5 & pin17
- DPort2 = pin4 & pin16

Analog Port = 1 pin (x1 Port)
- APort = pin0

I2C Port = 2 pin (x1 Port)
- I2CPort = pin22 && pin21

APスイッチ = 1 pin
- APPort = pin18

フルカラーLED = 3 pin
- pin27
- pin14
- pin12

電源→モジュール = 1 pin
- pin25

[image]

yam123, 2017-06-27:
Digital Port = 2 pin (x1 Port)
- DigiPort = pin0 & pin2

Analog(Digital) Port = 2 pin (x1 Port)
- AnaPort = TOUT & pin16

I2C Port = 2 pin (x2 Port)
- I2CPort1 = pin4 && pin5
- I2CPort2 = pin4 && pin5

APスイッチ = 1 pin
- APPort = pin14

フルカラーLED = 2 pin
- pin13
- pin15

電源→モジュール = 1 pin
- pin12

[image]

closed 2017-06-28 by yam123