HAL-RO-Developer/iot_plat_doc #27: サーボモーター制御

Komei1009, 2017-06-14:
# 背景
サーボモーター制御行うため
# 目的
サーボモーター制御の作品制作時に容易にするため
# 対応内容
ソースコード添付(コメントは可能な限り細かく)
# 期日
6/28

RikuHashimoto, 2017-06-21:
回路は
サーボモータのvccをESP8266のvin
GNDをGND
信号線をD1
にそれぞれ繋いでます

RikuHashimoto, 2017-06-21:
回転にかかる時間を計測した結果(マイクロ秒)
0度から45度の場合を１０回計測した結果
26
26
27
26
27
26
27
26
27
26
https://github.com/HAL-RO-Developer/iot_plat_doc/tree/master/work/hashimoto/servo

RikuHashimoto, 2017-06-21:
#include<Servo.h>

Servo servo1;

void setup() {
  Serial.begin(115200);
    servo1.attach(5); //ESP8266ボードのD1をサーボ用に設定
    
}

void loop() {
  unsigned long t1;
  unsigned long t2;
  //0→45→90→135→180の順で回す
   servo1.write(0);
    delay(1000);
    t1 = micros();//測定開始
    servo1.write(45);
    t2 = micros();//測定終了
    Serial.println(t2-t1);//表示
    delay(1000);
    servo1.write(90);
    delay(1000);
    servo1.write(135);
    delay(1000);
    servo1.write(180);
    delay(1000);
}

closed 2017-06-22 by makki0205