HAL-RO-Developer/iot_plat_doc #28: Groveモジュールの調査(RGB LED)

Komei1009, 2017-06-14:
# 背景
Groveモジュールを使用するため
# 目的
Grovejモジュールの仕様を明確にするため
# 対応内容
RGB LEDの応答時間、制御できる制度の調査(証拠添付)
# 期日
6/28

RikuHashimoto, 2017-06-26:
中間報告
まずこのモジュールを使うには下記URLからライブラリをインストールし、arduinoIDEのlibraryファイルに入れる必要があります。
http://wiki.seeedstudio.com/wiki/File:Chainable_LED_Library.zip

RikuHashimoto, 2017-06-26:
#include <ChainableLED.h> /* http://wiki.seeedstudio.com/wiki/File:Chainable_LED_Library.zip からインストールする*/
unsigned long t1;
unsigned long t2;
ChainableLED led(5, 4, 1);/*インスタンスを生成(引数は左から順に、クロックポート（モジュールの黄色の線を繋いでるポート)、データポート（モジュールの白線を繋いでるポート)、LED接続数を設定します。*/
void setup()
{
  Serial.begin(115200);
}
void loop(){
 t1 = micros();
 led.setColorRGB(0, 255, 0,0); /*引数は左からLED番号(モジュール上のJ1が0),Red0～255,Green0～255,Blue0～255*/
  t2 = micros();
  Serial.println(t2-t1);//表示
 delay(1000);
 led.setColorRGB(0, 0, 255, 0); 
 delay(1000);
led.setColorRGB(0, 0, 0,255); 
delay(1000);
}

応答速度は4177μs
回路はモジュールの2つあるコネクタの内、J1って書いている方を使います
配線はモジュールの赤(V)をマイコンのvin
黒(G)をGND
黄色(C1)をD1(GPIO5)
白色(D1)をD2(GPIO4)
に繋ぐ
RGBそれぞれ0から255の256段階で表現できます。

RikuHashimoto, 2017-06-28:
ChainableLED::ChainableLED(byte clk_pin, byte data_pin, byte number_of_leds) :
    _clk_pin(clk_pin), _data_pin(data_pin), _num_leds(number_of_leds)
{
    pinMode(_clk_pin, OUTPUT);
    pinMode(_data_pin, OUTPUT);
  
    _led_state = (byte*) calloc(_num_leds*3, sizeof(byte));

    for (byte i=0; i<_num_leds; i++)
        setColorRGB(i, 0, 0, 0);
}

ChainableLED::~ChainableLED()
{
    free(_led_state);
}

// --------------------------------------------------------------------------------------

void ChainableLED::clk(void)
{
    digitalWrite(_clk_pin, LOW);
    delayMicroseconds(_CLK_PULSE_DELAY); 
    digitalWrite(_clk_pin, HIGH);
    delayMicroseconds(_CLK_PULSE_DELAY);   
}

void ChainableLED::sendByte(byte b)
{
    // Send one bit at a time, starting with the MSB
    for (byte i=0; i<8; i++)
    {
        // If MSB is 1, write one and clock it, else write 0 and clock
        if ((b & 0x80) != 0)
            digitalWrite(_data_pin, HIGH);
        else
            digitalWrite(_data_pin, LOW);
        clk();

        // Advance to the next bit to send
        b <<= 1;
    }
}
 
void ChainableLED::sendColor(byte red, byte green, byte blue)
{
    // Start by sending a byte with the format "1 1 /B7 /B6 /G7 /G6 /R7 /R6"
    byte prefix = B11000000;
    if ((blue & 0x80) == 0)     prefix|= B00100000;
    if ((blue & 0x40) == 0)     prefix|= B00010000; 
    if ((green & 0x80) == 0)    prefix|= B00001000;
    if ((green & 0x40) == 0)    prefix|= B00000100;
    if ((red & 0x80) == 0)      prefix|= B00000010;
    if ((red & 0x40) == 0)      prefix|= B00000001;
    sendByte(prefix);
        
    // Now must send the 3 colors
    sendByte(blue);
    sendByte(green);
    sendByte(red);
}

void ChainableLED::setColorRGB(byte led, byte red, byte green, byte blue)
{
    // Send data frame prefix (32x "0")
    sendByte(0x00);
    sendByte(0x00);
    sendByte(0x00);
    sendByte(0x00);
    
    // Send color data for each one of the leds
    for (byte i=0; i<_num_leds; i++)
    {
        if (i == led)
        {
            _led_state[i*3 + _CL_RED] = red;
            _led_state[i*3 + _CL_GREEN] = green;
            _led_state[i*3 + _CL_BLUE] = blue;
        }
                    
        sendColor(_led_state[i*3 + _CL_RED], 
                  _led_state[i*3 + _CL_GREEN], 
                  _led_state[i*3 + _CL_BLUE]);
    }

    // Terminate data frame (32x "0")
    sendByte(0x00);
    sendByte(0x00);
    sendByte(0x00);
    sendByte(0x00);
}

RikuHashimoto, 2017-06-28:
#ifndef __ChainableLED_h__
#define __ChainableLED_h__

#include "Arduino.h"

#define _CL_RED             0
#define _CL_GREEN           1
#define _CL_BLUE            2
#define _CLK_PULSE_DELAY    20

class ChainableLED
{
public:
    ChainableLED(byte clk_pin, byte data_pin, byte number_of_leds);
    ~ChainableLED();
    
    void setColorRGB(byte led, byte red, byte green, byte blue);
    void setColorHSB(byte led, float hue, float saturation, float brightness);

private:
    byte _clk_pin;
    byte _data_pin;
    byte _num_leds; 

    byte* _led_state;
    
    void clk(void);
    void sendByte(byte b);
    void sendColor(byte red, byte green, byte blue);
};

#endif

closed 2017-06-30 by makki0205