HAL-RO-Developer/iot_plat_doc #32: Groveモジュールの調査(ブザー)

Komei1009, 2017-06-14:
# 背景
Groveモジュールを使用するため
# 目的
Groveモジュールの仕様を明確にするため
# 対応内容
- ブザーの応答時間
- 制御できる精度の調査(証拠添付)
# 期日
6/28

teruya002, 2017-06-16:
void setup() {
// put your setup code here, to run once: pinMode(6,OUTPUT);
}

void loop() {
// put your main code here, to run repeatedly: digitalWrite(6, HIGH);
delay(analogRead(0));
digitalWrite(6, LOW);
delay(analogRead(0));
}

teruya002, 2017-06-20:
[image]

teruya002, 2017-06-20:
↑の写真の配線でGROVEのブザーのテストに成功しました。
GROVEのワイヤーの黒がGND 赤が5V 白がDigital 7Pin  黄がDigital6Pin にそれぞれ接続しています。

show-t, 2017-06-20:
8,10行目でdelay(analogRead(0));とanalogRead(0)を引数に入れているのはなぜ？0番ピンは回路上で使っている感じもないので、挙動がわからないです。
ブザー音として確認できたのは1音だけですか？予測ですが上のプログラムだと正しく動いているとしても「プー」と同じ音しかならないのでは？ どういう結果が得られたのか書いてほしいです。
もう少し汎用的にして、ここの数値を変えればこう動くという根拠をのせてください。
その上で、音階域がどの程度出るか調査お願いします。

teruya002, 2017-06-22:
```sh
#define SW_PIN 14

unsigned long time1;
unsigned long time2;
unsigned long result;

#define BUZZER_PIN               6            /* sig pin of the buzzer */

int length = 15;         /* the number of notes */
char notes[] = "ccggaagffeeddc ";
int beats[] = { 1, 1, 1, 1, 1, 1, 2, 1, 1, 1, 1, 1, 1, 2, 4 };
int tempo = 300;

void setup()
{
  Serial.begin(115200);

  pinMode(SW_PIN,INPUT);
    /* set buzzer pin as output */
    pinMode(BUZZER_PIN, OUTPUT);
}

void loop()
{
    if(digitalRead(SW_PIN) == HIGH){
    time1 = micros();
    while(digitalRead(SW_PIN) == HIGH){}
  }else{
    time2 = micros();
    result = time2 - time1;
    Serial.println(result);
  }
    for(int i = 0; i < length; i++) {
        if(notes[i] == ' ') {
            delay(beats[i] * tempo);
        } else {
            playNote(notes[i], beats[i] * tempo);
        }
        delay(tempo / 2);    /* delay between notes */
    }

}

/* play tone */
void playTone(int tone, int duration) {
    for (long i = 0; i < duration * 1000L; i += tone * 2) {
        digitalWrite(BUZZER_PIN, HIGH);
        delayMicroseconds(tone);
        digitalWrite(BUZZER_PIN, LOW);
        delayMicroseconds(tone);
    }
}

void playNote(char note, int duration) {
    char names[] = { 'c', 'd', 'e', 'f', 'g', 'a', 'b', 'C'};
    int tones[] = { 1915, 1700, 1519, 1432, 1275, 1136, 1014, 956 };

    // play the tone corresponding to the note name
    for (int i = 0; i < 8; i++) {
        if (names[i] == note) {
            playTone(tones[i], duration);
        }
    }
}
```

以上のコードで試してみてブザー音での[きらきら星]の一部を鳴らせました。
上記コードのbeatは音を鳴らす時間（大きくすればそれだけ長く鳴る）550以上の値を入れるときらきら星の音がずれ始めました。
notes[]は下のほうのplaynoteと連動しており char name[ ] に記載している文字列を記載する事で順番を入れ替えれます
tones [ ] は音階で数字の上限で音が変化します(約7000以上の値を入れるときらきら星の音がずれ始めました)
tone [ ] の値を変えると↑の name [ ] も音が変わります。

音の最大制御は85dBです。
（参照URL http://wiki.seeed.cc/Grove-Buzzer/)

反応時間は約20です。

配線は前回と同じですが白色の部分を未接続でやりました。

show-t, 2017-06-22:
基本的なプログラムの流れは問題ないですが、回路が前回と同様ということは時間計測は正しく測れていないかと思います。もう一度回路とプログラムの確認をお願いします。（少しプログラムの修正も必要かもしれません）
それからもう一つ、playNote関数内のtonesのは適当な値でも大丈夫でしょうか？何か理由があって、1915や1700などという値が格納されているはずで、もっと言えば、namesとtonesが1対1に関連して成り立っています（cならば1915, fならば1432という具合です）。では、それらの数値はどのように決まっているのか。ヒントとして、圧電ブザーの仕組みを調べるといいです。音を鳴らす基本的なところなのでこれは技術検証というより照屋自身の知識として調査してください。

teruya002, 2017-06-22:
```sh

unsigned long time1=0;
unsigned long time2=0;

#define BUZZER_PIN               6            /* sig pin of the buzzer */

int length = 15;         /* the number of notes */
char notes[] = "ccggaagffeeddc ";
int beats[] = { 1, 1, 1, 1, 1, 1, 2, 1, 1, 1, 1, 1, 1, 2, 4 };
int tempo = 300;

void setup()
{
  Serial.begin(9600);
    /* set buzzer pin as output */
    pinMode(BUZZER_PIN, OUTPUT);
}

void loop()
{
    time1 = micros();
    for(int i = 0; i < length; i++) {
        if(notes[i] == ' ') {
            delay(beats[i] * tempo);
        } else {
            playNote(notes[i], beats[i] * tempo);
        }
        delay(tempo / 2);    /* delay between notes */
    }
                time2 = micros();
    Serial.println(time2-time1);
}

/* play tone */
void playTone(int tone, int duration) {
    for (long i = 0; i < duration * 1000L; i += tone * 2) {
        digitalWrite(BUZZER_PIN, HIGH);
        delayMicroseconds(tone);
        digitalWrite(BUZZER_PIN, LOW);
        delayMicroseconds(tone);
   
    }
}

void playNote(char note, int duration) {
    char names[] = { 'c', 'd', 'e', 'f', 'g', 'a', 'b', 'C'};
    int tones[] = { 1915, 1700, 1519, 1432, 1275, 1136, 1014, 956 };

    // play the tone corresponding to the note name
    for (int i = 0; i < 8; i++) {
        if (names[i] == note) {
            playTone(tones[i], duration);
        }
    }
}
```
プログラムに若干の変更を加えました。
これで調べていったところだろ
80~100の間くらいです。

toneの数値についてですが圧電ブザーは発振回路を内蔵しており周波数を高音圧で発するものという記述を見つけました。
http://www.murata.com/ja-jp/products/sound/sounder/pkb24spch
故にtoneの数値は各々の[周波数]の値だと思います。

closed 2017-06-25 by show-t