HAL-RO-Developer/iot_plat_doc #33: Groveモジュールの調査(赤外線送信器・受信器)

Komei1009, 2017-06-14:
# 背景
Groveモジュールを使用するため
# 目的
Groveモジュールの仕様を明確にするため
# 対応内容
- 赤外線送信器・受信器の応答時間
- 制御できる精度の調査(証拠添付)
# 期日
6/28

hiyanaka, 2017-06-22:
- 共通
赤 Vcc(5v or 3.3v)
黒 GND
白 NC(未接続)
- 赤外線送信モジュール
黄 Tx(送信)
- 赤外線受信モジュール
黄 Rx(受信)

hiyanaka, 2017-06-22:
# 受信モジュール
```c
#define PIN 4  // Rx
void setup() {
  pinMode(PIN, INPUT);
  Serial.begin(115200);
}

void loop() {
  int pin = digitalRead(PIN);
  Serial.println(pin);
}
```
信号があるときは 0、信号がないときは 1 が表示されるテストコード

- データシート
http://www.vishay.com/docs/82491/tsop382.pdf

hiyanaka, 2017-06-22:
# 送信モジュール
```c
#define PIN 4  // Tx
#define LED 16 // 動作確認用
 
// テスト信号
unsigned int data[] = {804,383,60,40,60,41,59,136,60,40,57,43,60,135,60,135,58,136,60,383,
                       54,141,59,135,60,40,60,40,60,41,60,40,60,41,60,40,59,2022};
int last = 0;
unsigned long us = micros();
 
// セットアップ
void setup() {
  pinMode(PIN, OUTPUT);
  pinMode(LED, OUTPUT);
}
 
// dataから信号を送信
void sendSignal() {
  int dataSize = sizeof(data) / sizeof(data[0]);
  for (int cnt = 0; cnt < dataSize; cnt++) {
    unsigned long len = data[cnt]*10;  // dataは10us単位でON/OFF時間を記録している
    unsigned long us = micros();
    do {
      digitalWrite(ir_out, 1 - (cnt&1)); // cntが偶数なら赤外線ON、奇数ならOFFのまま
      delayMicroseconds(8);  // キャリア周波数38kHzでON/OFFするよう時間調整
      digitalWrite(PIN, 0);
      delayMicroseconds(7);
    } while (long(us + len - micros()) > 0); // 送信時間に達するまでループ
  }
}

// メイン
void loop() {
  digitalWrite(LED, HIGH);
  sendSignal();
  digitalWrite(LED, LOW);
  delay(1000);
}
```
1秒間隔で指定した信号を送るテストコード

closed 2017-06-25 by show-t