HAL-RO-Developer/iot_plat_doc #35: Arduinoの処理時間計測プログラム

Komei1009, 2017-06-19:
# 背景
各モジュールの応答時間を知る必要があるため
# 目的
時間を容易に測れる様にするため
# 対応内容
対応時間を計測するプログラムを添付
# 期日
6/21

Komei1009, 2017-06-21:
``` c++
#define SW_PIN 14

unsigned long time1;
unsigned long time2;
unsigned long result;

void setup() {
  Serial.begin(115200);

	pinMode(SW_PIN,INPUT);
}

void loop() {

  
	if(digitalRead(SW_PIN) == HIGH){
		time1 = micros();
		while(digitalRead(SW_PIN) == HIGH){}
	}else{
		time2 = micros();
		result = time2 - time1;
    Serial.println(result);
	}
  
}

```

makki0205, 2017-06-26:
```c++
#define SW_PIN 14

unsigned long time1;
unsigned long time2;
unsigned long result;

void setup() {
  Serial.begin(115200);

	pinMode(SW_PIN,INPUT);
}

void loop() {

  
	if(digitalRead(SW_PIN) == HIGH){
               timer();
		while(digitalRead(SW_PIN) == HIGH){}
	}else{
		result = timer();
    Serial.println(result);
	}
  
}
int timer (){
static int beforeTime = 0;
int nowTime = micros();
int resultTime = nowTime-beforeTime;
beforeTime = nowTime
return resultTime
}
```

closed 2017-06-26 by Komei1009