HAL-RO-Developer/iot_plat_doc #40: GROVEモジュールを1PINで制御できるか

yam123, 2017-06-22:
# 背景
ESP8266のPINを余らせるため。
# 目的
1PINで制御できれば2PIN余ることでフルカラーLEDと電源→モジュール間のスイッチができる
# 対応内容
GROVEモジュールが1PINで制御させよう！
# 期日
06/22

teruya002, 2017-06-22:
[image]

上の写真の様な状態を指しているのならばこれでも制御は可能でした。
黒 GND
赤 Vcc(5V)
白 未接続
黄 Digital 6Pin
上記配線でもブザーのプログラムは問題なく動きました。

RikuHashimoto, 2017-06-22:
音センサと光センサは1ピンと電源だけで動きます

hiyanaka, 2017-06-22:
赤外線送信・受信モジュールともに1pinのみで制御可能
- 共通
赤 Vcc(5v or 3.3v)
黒 GND
白 NC(未接続)
- 赤外線送信モジュール
黄 Tx(送信)
- 赤外線受信モジュール
黄 Rx(受信)

makki0205, 2017-06-23:
1PINにしてしまうと以下の問題が発生します
- uartが使えない
- 赤外線送信moduleと受信moduleはコネクトpinのは配置が違うためどちらかが使用不可になる
- アルコールセンサ等の諸々のモジュールが利用できなくなる

なので却下させていただきます。

それならデジタルPIN減らして
- I2C ×2
- デジタルPIN ×1
- アナログPIN ×1

closed 2017-06-27 by yam123