HAL-RO-Developer/iot_plat_doc #46: 3DCAD図面作成

Komei1009, 2017-06-29:
# 背景
デバイス外装の図面作成のため
# 目的

# 対応内容
- デバイス外装の3D図面作成
# 期日
7/7

teruya002, 2017-07-09:
3D図面が完成したので添付しておきます。
[3Dtest 図面 v0.pdf](https://github.com/HAL-RO-Developer/iot_plat_doc/files/1133959/3Dtest.v0.pdf)

teruya002, 2017-07-10:
デバイス外装の図面の試作が完成したので添付しておきます。
[外装試作モデル 図面 v１.pdf](https://github.com/HAL-RO-Developer/iot_plat_doc/files/1136313/v.pdf)

teruya002, 2017-07-10:
上記のモデル図に大まかなモジュール用の穴やバッテリー収納場所へのスリットの作成を行いました。
[外装モデル 図面 v0.pdf](https://github.com/HAL-RO-Developer/iot_plat_doc/files/1136659/v0.pdf)

teruya002, 2017-07-12:
https://drive.google.com/drive/folders/0B0QRiuJQID2banhzQ3lnb2tEc1k
2Dのpdfが不具合を起こしていたため別途で記載します。

teruya002, 2017-07-19:
https://drive.google.com/drive/folders/0B0QRiuJQID2banhzQ3lnb2tEc1k
バッテリーを置く場所を修正したものの2Dデータです。