HAL-RO-Developer/iot_plat_doc #59: マイコンのレジスタ一覧を作成

yam123, 2017-06-29:
# 背景

# 目的
ソフトウエアが作業できるため

# 対応内容
ポートのレジスタや初期値の値を調べてまとめる。
※2年のStuduinoのPortMap(配)みたいなやつ

# 期日
7/4

yam123, 2017-07-04:
[レジスタ一覧.xlsx](https://github.com/HAL-RO-Developer/iot_plat_doc/files/1121382/default.xlsx)

yam123, 2017-07-05:
[Google Drive](https://drive.google.com/drive/folders/0B-DrtusZsmgaWGdDR3Uwdlc1STA)
レジスタ一覧 の名前でGoogleDriveに上げました。
確認をお願いします

show-t, 2017-07-06:
# ESP8366 MEMORY_MAP
（参考）http://wiki.jackslab.org/ESP8266_Memory_Map

GPIO | ADDRESS | GPIO NAME | PIN
-- | -- | -- | --
  | 0x60000300 | GPIO_OUT | PIN_OUT
  | 0x60000304 | GPIO_OUT_W1TS | PIN_OUT_SET:HIGH
  | 0x60000308 | GPIO_OUT_W1TC | PIN_OUT_CLEAR
  | 0x6000030C | GPIO_ENABLE | PIN_DIR：入出力
  | 0x60000310 | GPIO_ENABLE_W1TS | PIN_DIR_OUTPUT
  | 0x60000314 | GPIO_ENABLE_W1TC | PIN_DIR_INPUT
  | 0x60000318 | GPIO_IN | PIN_IN
  | 0x6000031C | GPIO_STATUS |  
  | 0x60000320 | GPIO_STATUS_W1TS |  
  | 0x60000324 | GPIO_STATUS_W1TC |  
GPIO 0 | 0x60000328 | GPIO_PIN0 | GPIO0_U,PIN_0
  | 0x6000032C | GPIO_PIN1 | U0TXD_U
GPIO 2 | 0x60000330 | GPIO_PIN2 | GPIO2_U,PIN_2
  | 0x60000334 | GPIO_PIN3 | U0RXD_U
GPIO 4 | 0x60000338 | GPIO_PIN4 | GPIO4_U
GPIO 5 | 0x6000033C | GPIO_PIN5 | GPIO5_U
  | 0x60000340 | GPIO_PIN6 | SD_CLK_U
  | 0x60000344 | GPIO_PIN7 | SD_DATA0_U
  | 0x60000348 | GPIO_PIN8 | SD_DATA1_U
  | 0x6000034C | GPIO_PIN9 | SD_DATA2_U
  | 0x60000350 | GPIO_PIN10 | SD_DATA3_U
  | 0x60000354 | GPIO_PIN11 | SD_CMD_U
GPIO12 | 0x60000358 | GPIO_PIN12 | MTDI_U
GPIO13 | 0x6000035C | GPIO_PIN13 | MTCK_U
GPIO14 | 0x60000360 | GPIO_PIN14 | MTMS_U
GPIO15 | 0x60000364 | GPIO_PIN15 | MTDO_U
  | 0x60000368 | GPIO_SIGMA_DELTA |  
  | 0x6000036C | GPIO_RTC_CALIB_SYNC |  
  | 0x60000370 | GPIO_RTC_CALIB_VALUE

@yam123 調べたら一瞬で出てきたので載せておきます。

closed 2017-07-06 by show-t

yam123, 2017-07-06:
[Google Drive レジスタ一覧](https://drive.google.com/drive/folders/0B-DrtusZsmgaWGdDR3Uwdlc1STA)