HAL-RO-Developer/iot_plat_doc #60: 降圧昇圧のテスト

yam123, 2017-06-30:
# 背景
[](http://eleshop.jp/shop/g/gF8S121/)
リコー ステップアップDC-DCコンバータIC/3Vの電子パーツを使って降圧できるか

# 目的
降圧昇圧回路を作らなくてもよくなる。

# 対応内容
電子パーツが狙った値で降圧してくれたらOK

- できた場合：同じ物の昇圧バージョンも購入。
- できなかった場合：降圧昇圧回路を組まないといけなくなる。

# 期日
7/3

yam123, 2017-07-04:
[使用するDC-DCコンバーター](http://eleshop.jp/shop/g/gDAN413/)

テスト用で部品を2個購入しました。
明日までにテストを終わらせたいと思います。

teruya002, 2017-07-13:
https://www.plexim.com/ja/academy/power-electronics/buck-conv
分圧を使いましたがこれでいけました。

teruya002, 2017-07-13:
[image]
降圧テスト（分圧）結果です

yam123, 2017-07-13:
[昇圧回路.asc](https://drive.google.com/drive/folders/0B-DrtusZsmgaWGdDR3Uwdlc1STA)

[image]

昇圧回路図です。

closed 2017-07-14 by yam123