HAL-RO-Developer/iot_plat_doc #63: JSONのParseでデータ有無の検索

show-t, 2017-07-03:
# 背景
JSONからオブジェクトにParseしたときにデータがない場合、アウトオブレンジを起こす可能性がある。それを回避するための仕組みが必要。

# 目的

# 対応内容
JSONからオブジェクトにParse（※ArduinoJSONライブラリを使用）
Parseしたデータがnullでないことを確認する。

# 期日
7/3

RikuHashimoto, 2017-07-03:
```c++
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);

  // シリアルの初期化待ち
  while (!Serial) { }

  // JSON用の固定バッファを確保。
  //この例では200文字分。
  StaticJsonBuffer<200> jsonBuffer;

  // char[]の形式でJSONの文字列を持つ。
  // char* 形式では文字列のコピーを保持してしまう
  char json[] =
      "{\"test\":\"test\",\"hoge\":hoge,\"data\":[48.756080,2.302038]}";

  // JsonObjectはJSONBufferに格納、パースされた結果を参照保持する。
  // データを参照する場合はこのJsonObjectを見に行く。
  // メモリはスコープから出た際に開放される。
  JsonObject& root = jsonBuffer.parseObject(json);

  // パースが成功したか確認。失敗したら終了
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  // データの取り出しは型変換
  const char* test = root["test"];
   const char* hoge= root["hoge"];
  double latitude = root["data"][0];
  double longitude = root["data"][1];

  // データの表示。.
   Serial.println();
  Serial.println(test);
  Serial.println(hoge);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {

}
```

RikuHashimoto, 2017-07-03:
test
hoge
48.756080
2.302038

RikuHashimoto, 2017-07-03:
keyがない場合はシリアルprintしても何も表示されない。
valueがない場合も何も表示されない
```c++
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);

  // シリアルの初期化待ち
  while (!Serial) { }

  // JSON用の固定バッファを確保。
  //この例では200文字分。
  StaticJsonBuffer<200> jsonBuffer;

  // char[]の形式でJSONの文字列を持つ。
  // char* 形式では文字列のコピーを保持してしまう
  char json[] =
      "{\"test\":}";

  // JsonObjectはJSONBufferに格納、パースされた結果を参照保持する。
  // データを参照する場合はこのJsonObjectを見に行く。
  // メモリはスコープから出た際に開放される。
  JsonObject& root = jsonBuffer.parseObject(json);

  // パースが成功したか確認。失敗したら終了
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  // データの取り出しは型変換
  
  const char* test = root["test"];
  const char* no= root["no"];
  // データの表示。

Serial.println();
Serial.println(test);
 Serial.println(no);
 Serial.println(test);
 
}
void loop() {
```

RikuHashimoto, 2017-07-04:
```c++
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
/*
char toSSID[] = "riku'siPhone"; //接続先SSID
char ssidPASSWD[] = "Rikuh9212"; //接続先のパスワード
char server[]="172.20.10.3";
*/
char server[]="192.168.10.4";
char toSSID[] = "aterm-87140d-g"; //接続先SSID
char ssidPASSWD[] = "2010d77774e09"; //接続先のパスワード
int a;
WiFiClient client;
void setup() {
  Serial.begin(115200);
 //WiFiクライアントモード設定
 WiFi.mode(WIFI_STA); 
  //WiFiの設定を入れる。IPは、デフォルトがDHCPのようで、IPが共有されるときはWiFiのアクセス情報だけでOK
  WiFi.begin(toSSID,ssidPASSWD);
  //接続されるまで待つ
WiFi.waitForConnectResult();
a = client.connect(server,2000); //サーバーに接続したら1を返す

  // シリアルの初期化待ち
  while (!Serial) { }
  Serial.println("");
  if(a == 1){
    Serial.println("OK");
  }else{
    Serial.println("NG");
    
  }
}

void loop() {
 if (client.available()) { //接続先のサーバからデータが来た
    char c = client.read();//格納
    Serial.print(c);//表示
  }
}
サーバからデータを受け取れました

RikuHashimoto, 2017-07-04:
```
192.168.10.7:16872が接続しました。
0
```

```
OK
{
  "hoge":"hoge"
}
```

RikuHashimoto, 2017-07-04:
```c++
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
/*
char toSSID[] = "riku'siPhone"; //接続先SSID
char ssidPASSWD[] = "Rikuh9212"; //接続先のパスワード
char server[]="172.20.10.3";
*/
char server[]="192.168.10.4";
char toSSID[] = "aterm-87140d-g"; //接続先SSID
char ssidPASSWD[] = "2010d77774e09"; //接続先のパスワード
int a;
WiFiClient client;
void setup() {
  Serial.begin(115200);
   // JSON用の固定バッファを確保。
  //この例では200文字分。
  StaticJsonBuffer<200> jsonBuffer;

  // char[]の形式でJSONの文字列を持つ。
  // char* 形式では文字列のコピーを保持してしまう
  char json[] =
      "{\"test\":\"test\",\"hoge\":hoge}";

  // JsonObjectはJSONBufferに格納、パースされた結果を参照保持する。
  // データを参照する場合はこのJsonObjectを見に行く。
  // メモリはスコープから出た際に開放される。
  JsonObject& root = jsonBuffer.parseObject(json);

  // パースが成功したか確認。失敗したら終了
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
 //WiFiクライアントモード設定
 WiFi.mode(WIFI_STA); 
  //WiFiの設定を入れる。IPは、デフォルトがDHCPのようで、IPが共有されるときはWiFiのアクセス情報だけでOK
  WiFi.begin(toSSID,ssidPASSWD);
  //接続されるまで待つ
WiFi.waitForConnectResult();
a = client.connect(server,2000); //サーバーに接続したら1を返す

  // シリアルの初期化待ち
  while (!Serial) { }
  Serial.println("");
  if(a == 1){
    Serial.println("OK");
   client.write("a");//送信
  }else{
    Serial.println("NG");
    
  }
}

void loop() {

}
```
サーバーに送信しました

RikuHashimoto, 2017-07-04:
```
192.168.10.7:27624が接続しました。
[受信]: a
```

closed 2017-07-04 by show-t